jiexiong2016/GCNv2_SLAM #2: Linking issue

binbin-xu, 2019-03-07:
Hi @jiexiong2016,

I have tried to compile your code but I got error in the final linking step:

> ../lib/libORB_SLAM2.so: undefined reference to `pangolin::Split(std::string const&, char)'
> ../lib/libORB_SLAM2.so: undefined reference to `pangolin::BindToContext(std::string)'
> ../lib/libORB_SLAM2.so: undefined reference to `DBoW2::FORB::toString(cv::Mat const&)'
> ../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)'
> ../lib/libORB_SLAM2.so: undefined reference to `DBoW2::FORB::fromString(cv::Mat&, std::string const&)'
> ../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreatePanel(std::string const&)'

Do you have any idea on how to fix it? I also tried to compile ORB-SLAM2 and that works fine. So I assume both DBOW2 and Pangolin are installed correctly.

Thank you for your help!

lericson, 2019-03-07:
Could you please post the output of `make VERBOSE=1`?

jiexiong2016, 2019-03-07:
Hi,

Which version of Pangolin and DBoW2 are you using? Have just tried with master branch of Pangolin, didn't manage to reproduce the linking error. We tested the code on several laptops, desktops and Jetson TX2. Our setup is as follows:

OS: Ubuntu 16.04
Cmake: 3.11.0
gcc version: 5.4.0
CUDA version: 9.2 and 10.0
Pangolin: master branch
DBoW: in thirdparty folder, it's slightly different with original ORB-SLAM2(added loading the vocabulary from a binary file).
libtorch: local built from pytorch v1.0.1 branch

binbin-xu, 2019-03-07:
Hi,

Thank you for your reply!
The output of `make VERBOSE=1` is:

> /usr/local/bin/cmake -E cmake_link_script CMakeFiles/rgbd_gcn.dir/link.txt --verbose=1
/usr/bin/c++   -Wall -O3 -march=native -std=c++11 -O3 -DNDEBUG  -rdynamic CMakeFiles/rgbd_gcn.dir/GCN2/rgbd_gcn.cc.o  -o ../GCN2/rgbd_gcn  -L/home/binbin/code/Tool/OpenNI2/Bin/x64-Release -Wl,-rpath,/home/binbin/code/Tool/GCNv2_SLAM/lib:/home/binbin/code/Tool/libtorch/cuda/lib:/usr/local/cuda/lib64:/usr/local/lib:/home/binbin/code/Tool/OpenNI2/Bin/x64-Release:/home/binbin/code/Tool/GCNv2_SLAM/Thirdparty/DBoW2/lib:/home/binbin/code/Tool/GCNv2_SLAM/Thirdparty/g2o/lib ../lib/libORB_SLAM2.so /home/binbin/code/Tool/libtorch/cuda/lib/libtorch.so -lcuda /usr/local/cuda/lib64/libnvrtc.so /usr/local/cuda/lib64/libnvToolsExt.so /usr/local/cuda/lib64/libcudart.so -Wl,--no-as-needed,/home/binbin/code/Tool/libtorch/cuda/lib/libcaffe2.so -Wl,--as-needed -Wl,--no-as-needed,/home/binbin/code/Tool/libtorch/cuda/lib/libcaffe2_gpu.so -Wl,--as-needed /home/binbin/code/Tool/libtorch/cuda/lib/libc10_cuda.so /home/binbin/code/Tool/libtorch/cuda/lib/libcaffe2.so /home/binbin/code/Tool/libtorch/cuda/lib/libc10.so -lpthread /usr/local/cuda/lib64/libcufft.so /usr/local/cuda/lib64/libcurand.so /usr/lib/x86_64-linux-gnu/libcudnn.so /usr/local/cuda/lib64/libculibos.a /usr/local/cuda/lib64/libcublas.so /usr/local/cuda/lib64/libcublas_device.a /usr/local/cuda/lib64/libcudart.so /usr/local/lib/libopencv_dnn.so.3.4.3 /usr/local/lib/libopencv_ml.so.3.4.3 /usr/local/lib/libopencv_objdetect.so.3.4.3 /usr/local/lib/libopencv_shape.so.3.4.3 /usr/local/lib/libopencv_stitching.so.3.4.3 /usr/local/lib/libopencv_superres.so.3.4.3 /usr/local/lib/libopencv_videostab.so.3.4.3 /usr/local/lib/libopencv_calib3d.so.3.4.3 /usr/local/lib/libopencv_features2d.so.3.4.3 /usr/local/lib/libopencv_flann.so.3.4.3 /usr/local/lib/libopencv_highgui.so.3.4.3 /usr/local/lib/libopencv_photo.so.3.4.3 /usr/local/lib/libopencv_video.so.3.4.3 /usr/local/lib/libopencv_videoio.so.3.4.3 /usr/local/lib/libopencv_imgcodecs.so.3.4.3 /usr/local/lib/libopencv_imgproc.so.3.4.3 /usr/local/lib/libopencv_viz.so.3.4.3 /usr/local/lib/libopencv_core.so.3.4.3 /usr/local/lib/libpangolin.so -lGL -lGLU -lGLEW -lSM -lICE -lX11 -lXext -lrt -lpthread -lavcodec -lavformat -lavutil -lswscale /usr/local/lib/librealsense.so -lOpenNI -lOpenNI2 -lpng -lz -ljpeg -ltiff -lIlmImf ../Thirdparty/DBoW2/lib/libDBoW2.so ../Thirdparty/g2o/lib/libg2o.so -ldl 
../lib/libORB_SLAM2.so: undefined reference to `pangolin::Split(std::string const&, char)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::BindToContext(std::string)'
../lib/libORB_SLAM2.so: undefined reference to `DBoW2::FORB::toString(cv::Mat const&)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)'
../lib/libORB_SLAM2.so: undefined reference to `DBoW2::FORB::fromString(cv::Mat&, std::string const&)'
../lib/libORB_SLAM2.so: undefined reference to `pangolin::CreatePanel(std::string const&)'
collect2: error: ld returned 1 exit status

For the version of Pangolin, I am also using the master branch version, which is up-to-date with the one in the github. For the DBoW2, I am using the one in thirdparty folder.

Also, my current setup is:
OS: Ubuntu 16.04
Cmake: 3.11.0-rc2
gcc version: 5.4.0 20160609
CUDA version: 9.0
Pangolin: master branch
DBoW: in thirdparty folder
libtorch: the one on the pytorch homepage https://download.pytorch.org/libtorch/cu90/libtorch-shared-with-deps-latest.zip

Thanks!

lericson, 2019-03-07:
Looks like your `libORB_SLAM2.so` isn't correctly linked. What is the output of `ldd lib/libORB_SLAM2.so`? Can you try rebuilding the project?

boonvon, 2019-03-17:
@binbin-xu 
I also encountered the same problem. Have you solved it? Can you tell me how to solve it?
[image]

boonvon, 2019-03-17:
@binbin-xu 
I also encountered the same problem. Have you solved it? Can you tell me how to solve it?

boonvon, 2019-03-17:
Hi @binbin-xu  
I also encountered the same problem. Have you solved it? Can you tell me how to solve it?Thanks!

filledlove, 2019-03-20:
Hello!
Have you solved the problem?

lericson, 2019-03-20:
@boonvon @filledlove What is the output of `ldd lib/libORB_SLAM2.so`? Can you try rebuilding the project?

jiexiong2016, 2019-03-20:
@binbin-xu @boonvon @filledlove I think the linking issue is due to the pre-built version of libtorch. Potentially due to the [conflict of C++ ABI](https://github.com/pytorch/pytorch/issues/13541). The error should be resolved by compiling the libtorch locally.

boonvon, 2019-03-20:
@jiexiong2016 @lericson Thank you for your reply!Since my own dependent library is too complicated, I rebuilt the system and built it from scratch. It has been successful. @jiexiong2016 Thank you for your work.  that is too awesome!

jiexiong2016, 2019-03-22:
Seems solved, close for now

closed 2019-03-22 by jiexiong2016

1171257311, 2019-11-19:
@binbin-xu friends,have you fix the problem?

1171257311, 2019-11-19:
i encountered the same question,I rebuilt the system and built it from scratch,did not work.

MinjieLin, 2020-01-02:
download Libtorch with cxx11 ABI from : 
 https://github.com/pytorch/pytorch/issues/17492#issuecomment-524692441

This works for me : )

vanguard478, 2022-10-25:
> download Libtorch with cxx11 ABI from : [pytorch/pytorch#17492 (comment)](https://github.com/pytorch/pytorch/issues/17492#issuecomment-524692441)
> 
> This works for me : )

Hi, 
I am trying to compile Libtorch from source, but unable to do so. Can you please tell me the exact libtorch version you used to build this repo? Is from the nightly version or the stable one?