jiexiong2016/GCNv2_SLAM #3: some problem when operating

dongdingcheng, 2019-03-14:
How can I solve the problem like this？
[image]

dongdingcheng, 2019-03-14:
[image]

jiexiong2016, 2019-03-14:
Hi,
Seems to be a loading error due to the path of GCNv2 model. The error should be be fixed if you run './rgbd_gcn' in folder GCN2 as the "run.sh" we provided. Alternatively, you could also provide the model path by using the environment variable "GCN_PATH" : https://github.com/jiexiong2016/GCNv2_SLAM/blob/75cb4e6862fd43e52a6cd176b25dbb211db629b5/src/GCNextractor.cc#L216
Besides, the vocabulary for GCNv2 should be "GCNvoc.bin".

dongdingcheng, 2019-03-21:
@jiexiong2016 Thank you for answering my questions. I having figured out the problem. Its something wrong about the CUDA which my version is CUDA 9.0. Then I upgrade to the CUDA 9.2. I can run the operation. It is an excellent work!!!
And then, I wonder how to train my own vocabulary with my own network. Could you tell me some details?

jiexiong2016, 2019-03-21:
Glad you solve the problem. Have just committed an example under GCNv2: GCNv2_SLAM/GCN2/create_voc.cc
The path to model and dataset should be modified to your own ones.      https://github.com/jiexiong2016/GCNv2_SLAM/blob/13fadb0d27599182155602619b64777ee8170c90/GCN2/create_voc.cc#L128

dongdingcheng, 2019-03-25:
Thank you, I will try my own model~

chaiein, 2019-05-17:
> Hi,
> Seems to be a loading error due to the path of GCNv2 model. The error should be be fixed if you run './rgbd_gcn' in folder GCN2 as the "run.sh" we provided. Alternatively, you could also provide [...]
Path to vocabulary should be GCNvoc.bin or ORBvoc.txt? 
 Program breaks at "module = torch::jit::load(net_fn); "
net_fn value is gcn2.pt
How to solve this issue

jiexiong2016, 2019-05-22:
It shouldn't have loading error if you run the "run.sh" from GCN2 folder. I can only suggest to check the compilation and installation of pytorch (and its dependencies) since the problem is likely related to these.

match08, 2019-05-28:
> Glad you solve the problem. Have just committed an example under GCNv2: GCNv2_SLAM/GCN2/create_voc.cc
> The path to model and dataset should be modified to your own ones.
> 
> https://github.com/jiexiong2016/GCNv2_SLAM/blob/13fadb0d27599182155602619b64777ee8170c90/GCN2/create_voc.cc#L128

no “saveToBinaryFile” function

jiexiong2016, 2019-05-28:
Have added the function

cumtchenchang, 2019-11-28:
Please forgive me not to open a new issue, my error is very related to this package. When I use create_voc.cc to train vocabulary, I encountered an error:
/chenchang1/data/CityCentre_test/0010.jpg
... Extraction done!
> Creating vocabulary. May take some time ...
Segmentation fault (core dumped)
There are only ten images in my folder, this error is about "voc.create(features);"
Can you solve this problem? Thank you, sincerely.