jiexiong2016/GCNv2_SLAM #4: A small question about paper

kxhit, 2019-03-21:
Thanks for sharing this nice work!
[image]

I have a question about the expression "if abs(xnj − xi) < c " at line 4 in the Algorithm 1 in paper "GCNv2: Efficient Correspondence Prediction for Real-Time SLAM, J. Tang, L. Ericson, J. Folkesson and P. Jensfelt, in arXiv:1902.11046, 2019". 
I think the "<" should be ">", because we want to tolerate the noise caused by imperfect camera calibration and ground truth. So we will not use points within "c" threshold (e.g c = 5 pixel) as negative samples, which are very likely to be the bias caused by noise.
Am I correct? Is this a typo mistake or am I getting it wrong? Waiting for your reply! Thanks!

jiexiong2016, 2019-03-21:
@kxhit Sorry for the mistake! This is a typo, it should be ">". Thank a lot for pointing out!

kxhit, 2019-03-22:
@jiexiong2016 Thank you for your prompt reply! Closed.

closed 2019-03-22 by kxhit