jiexiong2016/GCNv2_SLAM #5: Need some advice on network training losses

Merical, 2019-03-21:
Hi. I'm working on a program to extract keypoints and descriptors with pytorch. I'm really interested in your work and try to reimplement it in python. But I'm struggling with the loss function. May I ask for some advice or demo codes to help me complete the loss function for keypoints and descriptors? Thanks~

jiexiong2016, 2019-03-22:
Hi, thank you for your interest in GCNv2! The paper is currently under review, we plan to release the training code after the review process is completed.

Sincerely,
Jiexiong

ChLee98, 2019-09-27:
> Hi. I'm working on a program to extract keypoints and descriptors with pytorch. I'm really interested in your work and try to reimplement it in python. But I'm struggling with the loss function. May [...]

Hi, have you completed the loss function for keypoints and descriptors ? I also have some problems about the loss function.

match08, 2019-12-26:
publish traking GCN codes？

liandexu186, 2020-04-10:
> Hi, have you completed the loss function for keypoints and descriptors ? I also have some problems about the loss function.

me too