jiexiong2016/GCNv2_SLAM #9: The trajectory results are different with them in paper

xin-xiner, 2019-04-17:
We use CUDA10 & opencv4 & torch1.0 to build you project. 
Then the run.sh with gnc2_aug.pt is used.  The ATE on TUM fr1_desk, fr1_desk2 are nearly 0.4 m. 
I'm not sure where the problem is. Is there any suggestion?

jiexiong2016, 2019-04-17:
Hi, I have just built/tested with a fresh clone from master branch. I didnt have any issue reproducing the results in paper using gcn2_aug.pt on fr1_desk. The ATE I got using TUM evaluation tool is: 0.0289m. The trajectories are attached as follows: 
[CameraTrajectory.txt](https://github.com/jiexiong2016/GCNv2_SLAM/files/3091451/CameraTrajectory.txt)
[groundtruth.txt](https://github.com/jiexiong2016/GCNv2_SLAM/files/3091452/groundtruth.txt)
[image]

Maybe double check the calibration file and changes you have made to the code (if there is any). The ATE error should be relatively small using the current master branch.

Besides, it is recommended to at least use libtorch 1.0.1 to avoid some performance issue.

xin-xiner, 2019-04-18:
I find where the problem is. In GCN project, the image size is half of it in TUM datset. So, all the intrinsic parameters should be half.

jiexiong2016, 2019-04-18:
Glad you solved the problem. Currently the input image is resized to 320x240 at : https://github.com/jiexiong2016/GCNv2_SLAM/blob/0958a92beef3bc45b78a22f8c3303f66ab242383/src/Tracking.cc#L230
It is mainly for allowing Jetson TX2 can run real-time with the gcn2_tiny.pt. As we currently set the grid for bilinear sampling the feature with size 320x240 inside the exported model, it is not supporting other resolutions for now.