jiexiong2016/GCNv2_SLAM #10: Error compiling pytorch under windows

cugcsu, 2019-04-23:
c:\users\administrator\pytorch\aten\src\aten\cpu\vec256\vec256_int.h(428): error C3861: “_mm256_extract_epi64”: 找不到标识符 (编译源文件C:\Users\Administrator\pytorch\build\aten\src\ATen\native\cpu\ReduceOpsKernel.cpp.AVX2.cpp) 

When I use the compiled libtorch, I get an error when using it under Windows:
"std":Unclear symbol(不明确的符号)，ie.std::vector
When I looked up the std statement, I found that pytorch defined many std namespaces, which caused the name to be duplicated.

cugcsu, 2019-04-23:
@jiexiong2016

jiexiong2016, 2019-04-23:
Hi @cugcsu , I'm afraid I have limited knowledge about how to compile libtorch in Windows. The first error you pasted seems to be SSE related and the latter one might related to: https://discuss.pytorch.org/t/compiler-report-std-ambiguous-symbol-error-under-visual-c-2017/39459 and https://github.com/pytorch/pytorch/pull/17863.
You probably have to redirect to pytorch repository and find whether it has a solution there.

cugcsu, 2019-04-24:
> Hi @cugcsu , I'm afraid I have limited knowledge about how to compile libtorch in Windows. The first error you pasted seems to be SSE related and the latter one might related to: https://discuss.pytorch.org/t/compiler-report-std-ambiguous-symbol-error-under-visual-c-2017/39459 and [pytorch/pytorch#17863](https://github.com/pytorch/pytorch/pull/17863).
> You probably have to redirect to pytorch repository and find whether it has a solution there.

Do you know how to use the avx2 instruction set? In the vs project, I know that I can use the avx2 instruction set by enabling the enhanced instruction set, but in the cmakelists file, or somewhere else?

jiexiong2016, 2019-04-24:
I think the flag for enabling avx2 is set in here: https://github.com/pytorch/pytorch/blob/84b275b70f73d5fd311f62614bccc405f3d5bfa3/cmake/Dependencies.cmake#L1196

Using the pre-built libtorch library can cause linking issue: https://github.com/pytorch/pytorch/blob/84b275b70f73d5fd311f62614bccc405f3d5bfa3/cmake/Dependencies.cmake#L1196

The C10 error is probably caused by different version of CUDA as mentioned in another issue:
https://github.com/jiexiong2016/GCNv2_SLAM/issues/3#issuecomment-472859235

chaiein, 2019-05-17:
> I think the flag for enabling avx2 is set in here: https://github.com/pytorch/pytorch/blob/84b275b70f73d5fd311f62614bccc405f3d5bfa3/cmake/Dependencies.cmake#L1196
> 
> Using the pre-built libtorch library can cause linking issue: [#2 (comment)](https://github.com/jiexiong2016/GCNv2_SLAM/issues/2#issuecomment-470580288)
> 
> The C10 error is probably caused by different version of CUDA as mentioned in another issue:
> [#3 (comment)](https://github.com/jiexiong2016/GCNv2_SLAM/issues/3#issuecomment-472859235)
I have checked as per the Dependencies.cmake at  Line1196. Still I couldnt solve this issue. What is the reason and how to solve this?

cugcsu, 2019-05-22:
> I think the flag for enabling avx2 is set in here: https://github.com/pytorch/pytorch/blob/84b275b70f73d5fd311f62614bccc405f3d5bfa3/cmake/Dependencies.cmake#L1196
> Using the pre-built libtorch [...]

I installed pytorch in another way:
First, download the pytorch you need from [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pytorch).
Then, using pip install torch-1.0.1+cuda101-cp37-cp37m-win_amd64.whl 
But if you import this package in python, you will get the following error:
>>> import torch
Traceback (most recent call last):
   File "<stdin>", line 1, in <module>
   File "D:\SoftWare\Python\lib\site-packages\torch\__init__.py", line 80, in <module>
     From torch._C import *
ImportError: DLL load failed: The specified module could not be found.
>>>
At this time visit the following link:https://anaconda.org/anaconda/intel-openmp/files
Go to the downloaded file directory intel-openmp-2019.0-117\Library\bin and copy libiomp5md.dll and libiompstubs5md.dll to the directory where Python.exe is located.

Norfo, 2019-10-09:
I installed pytorch using the installation description on this site https://pytorch.org/get-started/locally/. At the moment, I am facing error C2872. I turned on and off the /permissive- flag. But nothing helped. Has anyone been able to solve this problem?