jiexiong2016/GCNv2_SLAM #13: Support MobileNet to make it faster?

richard-chau, 2019-05-13:
Does the author try the Mobilenet design? How much speedup and will performance change much?

jiexiong2016, 2019-06-10:
Recently trained a new model with mobilenet_v2 as encoder. Havent quite tested it yet, probably you need to double check if it can runs in real-time in your case.
[gcn2_mobile2_320x240.zip](https://github.com/jiexiong2016/GCNv2_SLAM/files/3273414/gcn2_mobile2_320x240.zip)

richard-chau, 2019-06-18:
Thanks! BTW, could you also share the network.py ?

jiexiong2016, 2019-06-18:
Updated in [Network.md](https://github.com/jiexiong2016/GCNv2_SLAM/blob/master/GCN2/Network.md). Besides, the mobilenet will not help to improve the speed since both GCNv2 and GCNv2_tiny are smaller than it and without batch normalization (could be good for performance though).