jiexiong2016/GCNv2_SLAM #19: 特征获取不错，但误差和ORB对比还很大

match08, 2019-05-28:
特征获取不错，但误差和ORB对比还很大。你可以插入一个opegl 3d 物体和orb的数据运行效果对比。

jiexiong2016, 2019-05-28:
Hi,
Thanks for your interests. GCNv2 is designed for generating reliable features under movement between frames, it is optimized for VO/SLAM purpose only and is a fairly small network current takes only 320x240 resolution image (small resolution is for real time performance on Jetson TX2). For generic/semantic object matching, it is beyond extent of this work. 
There are various related methods have demonstrated that deep learning can outform classical methods in case of generic object matching. But they usuall cannot do the inference in realtime and/or work on par with classical method on position estimation. We are planing to add semantic information and model the invariant transform directly in our future work. It likely needs a redesign of the network(deeper or wider) to extract higher level rather than only the low level features.
Besides, if you are interested in applying your own network, it is recommended to use a common cnn barebone and trained with higher resolution image since it has better capacity and local details. But still there will alway have trade-off between the performances of different applications.

cugcsu, 2019-05-29:
@jiexiong2016，Hello, can you provide training code now, I want to train my own sample.

jiexiong2016, 2019-05-29:
Hi, The paper is currently under review, we plan to release the training code after the review process is completed.

jiexiong2016, 2019-06-10:
Closed for now. In addition to my previous reply, you probably also need to check if the tracking is actually running in real-time on your computer.

closed 2019-06-10 by jiexiong2016