jiexiong2016/GCNv2_SLAM #24: 你好，不知道是我的pytorch装错了还是路径有问题？

450932894, 2019-07-02:
CMake Error at CMakeLists.txt:48 (find_package):
  By not providing "FindTorch.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Torch", but
  CMake did not find one.

  Could not find a package configuration file provided by "Torch" with any of
  the following names:

    TorchConfig.cmake
    torch-config.cmake

  Add the installation prefix of "Torch" to CMAKE_PREFIX_PATH or set
  "Torch_DIR" to a directory containing one of the above files.  If "Torch"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/tang/GCNv2_SLAM/build/CMakeFiles/CMakeOutput.log".
make: *** 没有指明目标并且找不到 makefile。 停止。

450932894, 2019-07-02:
请问你这个时无GPU版的pytorch么

450932894, 2019-07-02:
@jiexiong2016

450932894, 2019-07-02:
我想问下pytorch的版本问题？ Pytorch C++ api(libtorch)     和 Download here: 
https://download.pytorch.org/libtorch/cu90/libtorch-shared-with-deps-latest.zip   是一个版本么

closed 2019-07-02 by 450932894

jiexiong2016, 2019-07-05:
这个应该是路径问题，如果想用最新的api可以试一下[nightly build version](https://download.pytorch.org/libtorch/nightly/cu90/libtorch-shared-with-deps-latest.zip)。 我装的pytorch是直接从mater brach local build的。

ChLee98, 2019-07-11:
> CMake Error at CMakeLists.txt:48 (find_package):
> By not providing "FindTorch.cmake" in CMAKE_MODULE_PATH this project has
> asked CMake to find a package configuration file provided by "Torch", but [...]

我也遇到了这个问题，请问您最后怎么解决的呢？

sunpeng981712364, 2020-04-02:
在CmakeList.txt 中行首加上

sunpeng981712364, 2020-04-02:
In my case, adding these two lines at the beginning of GCNv2_SLAM/CMakeLists.txt solved the problem. Please modify the /path_to_your_torch/ part.

set(Caffe2_DIR "/path_to_your_torch/pytorch/torch/lib/tmp_install/share/cmake/Caffe2/")
include_directories("path_to_your_torch/pytorch/torch/lib/include")