jiexiong2016/GCNv2_SLAM #26: I think there is something wrong with GCNextractor.cc?    I hope you can solve this problem

450932894, 2019-07-03:
[ 17%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
/home/tang/GCNv2_SLAM/src/GCNextractor.cc:262:11: error: #elif with no expression
/home/tang/GCNv2_SLAM/src/GCNextractor.cc: In function ‘void ORB_SLAM2::nms(cv::Mat, cv::Mat, std::vector<cv::KeyPoint>&, cv::Mat&, int, int, int, int, float, float)’:
/home/tang/GCNv2_SLAM/src/GCNextractor.cc:96:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < pts_raw.size(); i++)
                       ^
/home/tang/GCNv2_SLAM/src/GCNextractor.cc:107:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < pts_raw.size(); i++)
                       ^
/home/tang/GCNv2_SLAM/src/GCNextractor.cc:148:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i=0; i<select_indice.size(); i++)
                    ^
/home/tang/GCNv2_SLAM/src/GCNextractor.cc: In member function ‘void ORB_SLAM2::GCNextractor::operator()(cv::InputArray, cv::InputArray, std::vector<cv::KeyPoint>&, cv::OutputArray)’:
/home/tang/GCNv2_SLAM/src/GCNextractor.cc:269:22: error: ‘img_var’ was not declared in this scope
     inputs.push_back(img_var);
                      ^
CMakeFiles/ORB_SLAM2.dir/build.make:127: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/GCNextractor.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/GCNextractor.cc.o] Error 1
make[2]: *** 正在等待未完成的任务....
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

[GCNextractor.txt](https://github.com/jiexiong2016/GCNv2_SLAM/files/3354514/GCNextractor.txt)

450932894, 2019-07-03:
@jiexiong2016

iamwangyabin, 2019-07-05:
I also had this problem...

jiexiong2016, 2019-07-05:
My bad. There was a typo I made at https://github.com/jiexiong2016/GCNv2_SLAM/blob/63c309e1e8190d40be6126da00f8ceb42fb2d528/src/GCNextractor.cc#L262 when check the pytorch version. I have just fixed it, can you try again?

iamwangyabin, 2019-07-06:
@jiexiong2016  I had successfully run your program , it looks pretty good, I can run it realtime on my laptop! Do you consider release a ROS version ?

450932894, 2019-07-06:
I try again.But there are new problems. I installed it with 【sudo pip3 install torch torchvision】. Do I need to install libtouch again?
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_gcn.dir/build.make:168: recipe for target '../GCN2/rgbd_gcn' failed
make[2]: *** [../GCN2/rgbd_gcn] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/rgbd_gcn.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_gcn.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

450932894, 2019-07-06:
Hi，I try again.But there are new problems. I installed it with 【sudo pip3 install torch torchvision】. Do I need to install libtouch again? 
This my    cmake .. -DCMAKE_BUILD_TYPE=Release -DTORCH_PATH=/usr/local/lib/python3.5/dist-packages/torch/share/cmake/Torch
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_gcn.dir/build.make:168: recipe for target '../GCN2/rgbd_gcn' failed
make[2]: *** [../GCN2/rgbd_gcn] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/rgbd_gcn.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_gcn.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

jiexiong2016, 2019-07-06:
I think this is related to the installation of libtorch. Check #2 for more details. I would recommend simply build the libtorch locally using the master branch.

450932894, 2019-07-09:
@iamwangyabin @jiexiong2016 I successfully ran the following three steps

Git clone -- recursive - b v1.0.1 https://github.com/pytorch/pytorch

CD pytorch && mkdir build && CD build

Python.. / tools/build_libtorch. Py

But it still reported an error.

Torch version shows 1.0.0
-- Caffe2: Header version is: 9.0
-- Found CUDNN: /usr/include  
-- Found cuDNN: v7.5.0  (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libcudnn.so)
-- Autodetected CUDA architecture(s):  6.1
-- Added CUDA NVCC flags for: -gencode;arch=compute_61,code=sm_61
-- Found torch: /home/tang/pytorch/torch/lib/libtorch.so  
-- Torch version is: 1.0.0
-- Configuring done
-- Generating done
I hope you can help me. This project is important to my research. My qq is 450932894.WeChat too.

450932894, 2019-07-09:
I did download pytorch 1.1.0.

star0w, 2021-06-23:
> I did download pytorch 1.1.0.

hello I have met the same problem,  can u remember how did u solve it?

Antony666, 2022-01-15:
> Hi，I try again.But there are new problems. I installed it with 【sudo pip3 install torch torchvision】. Do I need to install libtouch again? This my cmake .. -DCMAKE_BUILD_TYPE=Release [...]

ubuntu 16.04
pytorch 1.2.0
Hi! I also have this problem.how did you solve this problem?  thanks.