jiexiong2016/GCNv2_SLAM #28: how to get association file?

jinfagang, 2019-07-19:
I can not found such a file in downloaded data:

```
\./Datasets/rgbd_dataset_freiburg3_long_office_household/associations.txt

```

jinfagang, 2019-07-19:
I found a asociate.py buy generate associations.txt were not right:

```
1341847980.790000 -0.6832 2.6909 1.7373 0.0003 0.8617 -0.5072 -0.0145 1341847980.786879 depth/1341847980.786879.png 1341847980.786856 rgb/1341847980.786856.png
1341847980.820100 -0.6821 2.6914 1.7371 0.0003 0.8609 -0.5085 -0.0151 1341847980.822989 depth/1341847980.822989.png 1341847980.822978 rgb/1341847980.822978.png
1341847980.850000 -0.6811 2.6918 1.7371 0.0001 0.8610 -0.5084 -0.0159 1341847980.854690 depth/1341847980.854690.png 1341847980.854676 rgb/1341847980.854676.png
1341847980.890100 -0.6797 2.6921 1.7373 0.0000 0.8610 -0.5083 -0.0162 1341847980.890737 depth/1341847980.890737.png 1341847980.890728 rgb/1341847980.890728.png
1341847980.920000 -0.6785 2.6925 1.7375 -0.0003 0.8607 -0.5088 -0.0160 1341847980.922989 depth/1341847980.922989.png 1341847980.922978 rgb/1341847980.922978.png
1341847980.950000 -0.6775 2.6928 1.7374 -0.0001 0.8607 -0.5088 -0.0166 1341847980.954676 depth/1341847980.954676.png 1341847980.954645 rgb/1341847980.954645.png
```

Runs error:

```
\

Failed to load image at: ./Datasets/rgbd_dataset_freiburg3_long_office_household/-0.6832
Segmentation fault (core dumped)

```

jinzhenmu, 2019-07-19:
you can go to ORB-SLAM2(github)  and download this txt . this txt location at Examples/RGB-D

jinfagang, 2019-07-19:
@jinzhenmu  I think it's not a txt but a folder, there were no such file association.txt:

[image]

jinfagang, 2019-07-19:
ok, I got it, it's different enviroments

closed 2019-07-19 by jinfagang

jinfagang, 2019-07-19:
Ok, I make it worked, but the speed seems a little stuck on GTX1080ti 

[image]

jinzhenmu, 2019-07-19:
my  enviroments is cuda9.0 cudnn7.6.1  
computer is ASUS-FXJ50

sharoseali, 2020-06-04:
> Ok, I make it worked, but the speed seems a little stuck on GTX1080ti
> 
> [image]

HI @jinfagang @jinzhenmu ... I am trying to run the code but run.sh didn't show any output after runing the following command written in run.sh

`GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml ~/Datasets/rgbd_dataset_freiburg3_long_office_household/rgb ~/Datasets/rgbd_dataset_freiburg3_long_office_household/fr3_office_val.txt`
please help also I want to run it on my TX2 based car robot/ I need your guidance how to run it after getting live video stream from jet son camera instead of existing data-set images. 
Thanks in advance