jiexiong2016/GCNv2_SLAM #35: Problems with model.

Norfo, 2019-10-21:
Thanks for this project. I use Windows 10 and have some problems with the model. I described it in detail in the pytorch repository (https://github.com/pytorch/pytorch/issues/28276). For the problem with loading the model, a solution was proposed. It was necessary to slightly fix the model. At the moment, I am faced with the fact that the forward() method does not work correctly. Do you have any thoughts about this error? Could this be due to the fact that I'm using version 1.3 of the pytorch? Or maybe this is a model problem?

Currently I have the following error:
```
The above operation failed in interpreter, with the following stack trace:
at code/gcn2_480x640.py:43:8
  _6 = torch.unsqueeze(torch.select(_5, 1, 0), 1)
  _7 = torch.to(_6, 6, False, False)
  _8 = torch.slice(vu_1, 0, 0, 9223372036854775807, 1)
  _9 = torch.select(_8, 1, 0)
  _10 = torch.slice(vu_1, 0, 0, 9223372036854775807, 1)
  _11 = torch.select(_10, 1, 1)
  _12 = torch.to(_9, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  _13 = torch.to(_11, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  _14 = torch.unsqueeze(torch.index(det, [_12, _13]), 1)
  pts = torch.cat([_4, _7, _14], 1)
        ~~~~~~~~~ <--- HERE
  _15 = torch.slice(pts, 0, 0, 9223372036854775807, 1)
  input_20 = torch.neg(torch.select(_15, 1, 2))
  _16, inds_1 = torch.sort(input_20, -1, False)
  inds_2 = torch.to(inds_1, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  _17 = torch.index(pts, [inds_2])
  inds = torch.to(inds_2, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  vu = torch.index(vu_1, [inds])
  _18 = torch.slice(vu, 0, 0, 9223372036854775807, 1)
  _19 = torch.select(_18, 1, 0)
```

Luxcy, 2019-11-01:
I don't have GPU. I want to run it on CPU. I changed codes. 
I have a similar error when forward()  is running:
'
operation failed in interpreter:
  _3 = torch.unsqueeze(torch.select(_2, 1, 1), 1)
  _4 = torch.to(_3, 6, False, False)
  _5 = torch.slice(vu_1, 0, 0, 9223372036854775807, 1)
  _6 = torch.unsqueeze(torch.select(_5, 1, 0), 1)
  _7 = torch.to(_6, 6, False, False)
  _8 = torch.slice(vu_1, 0, 0, 9223372036854775807, 1)
  _9 = torch.select(_8, 1, 0)
  _10 = torch.slice(vu_1, 0, 0, 9223372036854775807, 1)
  _11 = torch.select(_10, 1, 1)
  _12 = torch.to(_9, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
        ~~~~~~~~ <--- HERE
  _13 = torch.to(_11, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  _14 = torch.unsqueeze(torch.index(det, [_12, _13]), 1)
  pts = torch.cat([_4, _7, _14], 1)
  _15 = torch.slice(pts, 0, 0, 9223372036854775807, 1)
  input_20 = torch.neg(torch.select(_15, 1, 2))
  _16, inds_1 = torch.sort(input_20, -1, False)
  inds_2 = torch.to(inds_1, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
  _17 = torch.index(pts, [inds_2])
  inds = torch.to(inds_2, dtype=4, layout=0, device=torch.device("cuda:0"), non_blocking=False, copy=False)
'
Is to(..,device=torch.device("cuda:0"), ...) in GCN2 Net?Thank you very much!

Norfo, 2019-11-01:
Hello, Luxcy! Just yesterday I solved my problem. The fact is that I have wrong paths for pytorch dlls in my assembly. I just copied all the libraries (such as in my case c_10, c10_cuda, caffe2_detectron_ops_gpu, caffe2_module_test_dynamic, caffe2_nvrtc, caffe2_observers, cudnn64_7, shm, torch, torch_python) to the my project folder, and it worked. I’m not sure that this will be a 100% solution to your problem, but it can somehow help.

Luxcy, 2019-11-01:
Ok, I get it! 
Thank you very much, Norfo!
I will try it. But is all the libraries copied to the GCNv2_SLAM dir ? or subdirectory?
I

Norfo, 2019-11-01:
Firstly, you should build GCNv2_SLAM  project (for example as a .lib file). Then you shuold create a project that use that lib. Finally, copy pytorch dlls into your project folder. Dont forget replace all cuda mentions in GCNv2_SLAM and .pt file.

Luxcy, 2019-11-01:
I didn't change the .pt file!
Now, It can run goodly.
Thank you very much!
Thanks!!！

Norfo, 2019-11-01:
Happy for you. Just in case. You can unzip the .pt file (it is just a simple zip archive). Open and edit the .py file in unzip directory. Then pack again using any archiver.

closed 2019-11-01 by Norfo

Luxcy, 2019-11-01:
You are a very good teacher!
I do what you say! 
Unzip and pack it .zip. It succeeds.

liubamboo, 2020-12-10:
@Norfo Thanks for your work. Do you know how to generate the pt file? train in pytorch project or C++ project?

kintzhao, 2022-10-09:
> Firstly, you should build GCNv2_SLAM project (for example as a .lib file). Then you shuold create a project that use that lib. Finally, copy pytorch dlls into your project folder. Dont forget replace all cuda mentions in GCNv2_SLAM and .pt file.

how to process the pytorch dlls  for cpu, can you give it in detail ?