jiexiong2016/GCNv2_SLAM #38: linking errors

1171257311, 2019-11-19:
this question was raised before(https://github.com/jiexiong2016/GCNv2_SLAM/issues/2),however ,according to the previous method,it is not solved on my computer,so i want to ask others ,did you encounter the similar question?and how to solve it?

`../lib/libORB_SLAM2.so: undefined reference to pangolin::Split(std::string const&, char)' ../lib/libORB_SLAM2.so: undefined reference to pangolin::BindToContext(std::string)'
../lib/libORB_SLAM2.so: undefined reference to DBoW2::FORB::toString(cv::Mat const&)' ../lib/libORB_SLAM2.so: undefined reference to pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)'
../lib/libORB_SLAM2.so: undefined reference to DBoW2::FORB::fromString(cv::Mat&, std::string const&)' ../lib/libORB_SLAM2.so: undefined reference to pangolin::CreatePanel(std::string const&)'
`
my setup:
ubuntu 16.04
gcc 5.4.0
g++ 5.4.0
cmake 3.11.3
pytorch 1.1.0
cuda 9.2

1171257311, 2019-11-21:
i have solved it，first you must install pytorch from source，dont install from Binaries，you should
follow this:   git clone --recursive -b v1.0.1 https://github.com/pytorch/pytorch
pay attention,v1.0.1.
(In china,the speed is very slow,be patient! And  9:00~10:00 everyday，the speed seem to be faster. )
and run :python ../tools/build_libtorch.py
you will see libtorch.so in pytorch/torch/lib/tmp_install/lib.
then ,you should modify build.sh  : -DTORCH_PATH=/home/***/pytorch/torch/lib/tmp_install/share/cmake/Torch.(a little different from origin directory)
you should see many .cmake files under this directory(/home/***/pytorch/torch/lib/tmp_install/share/cmake) ,including Caffe2 ,Torch.
Then ,run build.sh.No error!
if you have any question,tell me

closed 2019-11-21 by 1171257311

cumtchenchang, 2019-11-26:
@1171257311  I followed your steps, but still have error in the final.
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
[100%] Linking CXX executable ../GCN2/rgbd_gcn
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by //usr/lib/x86_64-linux-gnu/libavfilter-ffmpeg.so.5, may conflict with libopencv_core.so.3.3
../lib/libORB_SLAM2.so: undefined reference to `torch::jit::GraphExecutor::run(std::vector<c10::IValue, std::allocator<c10::IValue> >&)'
../lib/libORB_SLAM2.so: undefined reference to `torch::autograd::Variable::Impl::Impl(at::Tensor, bool, torch::autograd::Edge)'
../lib/libORB_SLAM2.so: undefined reference to `torch::jit::GraphExecutor::GraphExecutor(std::shared_ptr<torch::jit::Graph>, bool)'
../lib/libORB_SLAM2.so: undefined reference to `torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>)'
../lib/libORB_SLAM2.so: undefined reference to `torch::jit::Value::uniqueNameBase[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_gcn.dir/build.make:166: recipe for target '../GCN2/rgbd_gcn' failed
make[2]: *** [../GCN2/rgbd_gcn] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/rgbd_gcn.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_gcn.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

1171257311, 2019-11-26:
是使用这个指令安装的吗？ git clone --recursive -b v1.0.1 https://github.com/pytorch/pytorch
还有麻烦发一下你的配置，我的配置：
my setup:
ubuntu 16.04
gcc 5.4.0
g++ 5.4.0
cmake 3.11.3
pytorch 1.1.0
cuda 9.2

cumtchenchang, 2019-11-26:
是的 是按照git clone --recursive -b v1.0.1 https://github.com/pytorch/pytorch进行下载的，
然后python ../tools/build_libtorch.py，之后将-DTORCH_PATH=改成了我自己的路径。
ubuntu 16.04
gcc 5.4.0 
g++ 5.4.0
cmake 3.10.1
pytorch 0.4.1
cuda  9.0.176

cumtchenchang, 2019-11-26:
@jiexiong2016  请问作者之前遇到这种问题吗？不知道能不能解决一下，谢谢

1171257311, 2019-11-26:
这个问题倒是没见过，另外安装的是1.0.1，为什么你的变成了1.1.0，你的pytorch/torch/lib/tmp_install/lib目录下有libtorch.so 文件吗？

cumtchenchang, 2019-11-26:
@1171257311  pytorch是1.0.1 我的系统还有pip安装的pytorch，刚刚误把版本写成了那个。pytorch/torch/lib/tmp_install/lib目录下有libtorch.so 文件。
[image]

1171257311, 2019-11-26:
你的opencv版本是多少？

1171257311, 2019-11-26:
还有你的系统安装了使用pip安装的pytorch，有可能会冲突，建议删除，只保留一个通过源码安装的。

cumtchenchang, 2019-11-26:
我已经卸载了，opencv版本是