jiexiong2016/GCNv2_SLAM #39: GCN-SLAM runing only use RGB?

jinzhenmu, 2019-11-22:
The GCN-SLAM using the RGB-D. And it can runing only use RGB?

DreamWaterFound, 2021-08-09:
Yes.
You just need to hack some codes in `Tracking::GrabImageMonocular()` and overload `Frame` constructor with `GCNextractor`.

szgy66, 2021-12-09:
> Yes. You just need to hack some codes in `Tracking::GrabImageMonocular()` and overload `Frame` constructor with `GCNextractor`.

Can you tell the details?

leslieburke, 2022-07-19:
> > Yes. You just need to hack some codes in `Tracking::GrabImageMonocular()` and overload `Frame` constructor with `GCNextractor`.
> 
> Can you tell the details?

Hello，Did you solve it？

cyizhuo, 2023-11-15:
> > > 是的。您只需要修改一些代码并`Tracking::GrabImageMonocular()`使用.`` Frame``GCNextractor ``
> > 
> > 
> > 我目前正在破解Track.cc和Frame.cc，已经用了半天​​了，Tracking还是返回一个空的pose……
> 
> Did you succeed?

Yes, I'll post my code later, but my result is that gcn works too slow on pure CPU, and it also lost track from time to time. I think we have to train our own feature model to achive better result. So I'm back to good old orb_slam3.

cyizhuo, 2023-11-15:
> > > > > 是的。您只需要修改一些代码并`Tracking::GrabImageMonocular()`使用.` Frame`GCNextractor ``
> > > > 
> > > > 
> > > > 我目前正在破解Track.cc和Frame.cc，已经用了半天​​了，Tracking还是返回一个空的pose……
> > > 
> > > 
> > > Did you succeed?
> > 
> > 
> > Yes, I'll post my code later, but my result is that gcn works too slow on pure CPU, and it also lost track from time to time. I think we have to train our own feature model to achive better result. So I'm back to good old orb_slam3.
> 
> Thank you very much for your reply. When will the code be released?

https://github.com/cyizhuo/GCNv2_SLAM_modified