jiexiong2016/GCNv2_SLAM #41: create_voc error

cumtchenchang, 2019-11-26:
Thank you for your good work, I want to follow your create_voc.cc to train vocabulary, but I encounter an error.

[ 91%] Built target ORB_SLAM2
[ 95%] Linking CXX executable ../GCN2/create_voc
CMakeFiles/create_voc.dir/GCN2/create_voc.cc.o: In function `void std::__once_call_impl<std::_Bind_simple<torch::jit::script::Method::get_executor()::{lambda()#1} ()> >()':
create_voc.cc:(.text._ZSt16__once_call_implISt12_Bind_simpleIFZN5torch3jit6script6Method12get_executorEvEUlvE_vEEEvv[_ZSt16__once_call_implISt12_Bind_simpleIFZN5torch3jit6script6Method12get_executorEvEUlvE_vEEEvv]+0xb6): undefined reference to `torch::jit::GraphExecutor::GraphExecutor(std::shared_ptr<torch::jit::Graph>, bool)'
CMakeFiles/create_voc.dir/GCN2/create_voc.cc.o: In function `torch::jit::script::Method::defaultSchemaFor(torch::jit::script::Method const&)':
create_voc.cc:(.text._ZN5torch3jit6script6Method16defaultSchemaForERKS2_[_ZN5torch3jit6script6Method16defaultSchemaForERKS2_]+0x1b2): undefined reference to `torch::jit::Value::uniqueNameBase[abi:cxx11]() const'
CMakeFiles/create_voc.dir/GCN2/create_voc.cc.o: In function `main':
create_voc.cc:(.text.startup+0xdd): undefined reference to `torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>)'
create_voc.cc:(.text.startup+0x56e): undefined reference to `torch::autograd::Variable::Impl::Impl(at::Tensor, bool, torch::autograd::Edge)'
create_voc.cc:(.text.startup+0xf5c): undefined reference to `torch::jit::GraphExecutor::run(std::vector<c10::IValue, std::allocator<c10::IValue> >&)'
collect2: error: ld returned 1 exit status
CMakeFiles/create_voc.dir/build.make:166: recipe for target '../GCN2/create_voc' failed
make[2]: *** [../GCN2/create_voc] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/create_voc.dir/all' failed
make[1]: *** [CMakeFiles/create_voc.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Thank you, sincerely.

cumtchenchang, 2019-12-06:
In your code:
cv::Mat desc_mat(cv::Size(32, pts.size(0)), CV_8UC1, desc.data<unsigned char>());
Dose you use 32-D trained descriptors?

liubamboo, 2020-12-08:
@cumtchenchang Thanks for your question. Have you solved the problem?