jiexiong2016/GCNv2_SLAM #47: #error C++14 or later compatible compiler is required to use ATen

logindiamant, 2020-07-22:
Dear colleagues, please tell, how to solve this problem?
...
-- Build files have been written to: /home/gleb/GCNv2_SLAM/build
[  4%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[  8%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[ 13%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
[ 17%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
In file included from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3:0,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:25,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:24,
                 from /home/gleb/GCNv2_SLAM/include/LocalMapping.h:24,
                 from /home/gleb/GCNv2_SLAM/src/LocalMapping.cc:21:
/home/gleb/pytorch/torch/include/ATen/ATen.h:4:2: error: #error C++14 or later compatible compiler is required to use ATen.
 #error C++14 or later compatible compiler is required to use ATen.
  ^~~~~
In file included from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3:0,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:25,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:24,
                 from /home/gleb/GCNv2_SLAM/include/LoopClosing.h:24,
                 from /home/gleb/GCNv2_SLAM/src/LoopClosing.cc:21:
/home/gleb/pytorch/torch/include/ATen/ATen.h:4:2: error: #error C++14 or later compatible compiler is required to use ATen.
 #error C++14 or later compatible compiler is required to use ATen.
  ^~~~~
In file included from /home/gleb/pytorch/torch/include/c10/util/ArrayRef.h:19:0,
                 from /home/gleb/pytorch/torch/include/c10/core/MemoryFormat.h:5,
                 from /home/gleb/pytorch/torch/include/ATen/core/TensorBody.h:5,
                 from /home/gleb/pytorch/torch/include/ATen/Tensor.h:3,
                 from /home/gleb/pytorch/torch/include/ATen/Context.h:4,
                 from /home/gleb/pytorch/torch/include/ATen/ATen.h:9,
                 from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:25,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:24,
                 from /home/gleb/GCNv2_SLAM/include/LocalMapping.h:24,
                 from /home/gleb/GCNv2_SLAM/src/LocalMapping.cc:21:
/home/gleb/pytorch/torch/include/c10/util/C++17.h:24:2: error: #error You need C++14 to compile PyTorch
 #error You need C++14 to compile PyTorch
  ^~~~~
In file included from /home/gleb/pytorch/torch/include/c10/util/ArrayRef.h:19:0,
                 from /home/gleb/pytorch/torch/include/c10/core/MemoryFormat.h:5,
                 from /home/gleb/pytorch/torch/include/ATen/core/TensorBody.h:5,
                 from /home/gleb/pytorch/torch/include/ATen/Tensor.h:3,
                 from /home/gleb/pytorch/torch/include/ATen/Context.h:4,
                 from /home/gleb/pytorch/torch/include/ATen/ATen.h:9,
                 from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:25,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:24,
                 from /home/gleb/GCNv2_SLAM/include/LoopClosing.h:24,
                 from /home/gleb/GCNv2_SLAM/src/LoopClosing.cc:21:
/home/gleb/pytorch/torch/include/c10/util/C++17.h:24:2: error: #error You need C++14 to compile PyTorch
 #error You need C++14 to compile PyTorch
  ^~~~~
In file included from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3:0,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:29,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:24,
                 from /home/gleb/GCNv2_SLAM/include/FrameDrawer.h:25,
[... 93 lines cut ...]
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:29,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:24,
                 from /home/gleb/GCNv2_SLAM/include/FrameDrawer.h:25,
                 from /home/gleb/GCNv2_SLAM/include/Viewer.h:25,
                 from /home/gleb/GCNv2_SLAM/include/Tracking.h:28,
                 from /home/gleb/GCNv2_SLAM/include/System.h:30,
                 from /home/gleb/GCNv2_SLAM/src/System.cc:23:
/home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/all.h:4:2: error: #error C++14 or later compatible compiler is required to use PyTorch.
 #error C++14 or later compatible compiler is required to use PyTorch.
  ^~~~~
In file included from /home/gleb/pytorch/torch/include/c10/util/Exception.h:5:0,
                 from /home/gleb/pytorch/torch/include/c10/core/Device.h:5,
                 from /home/gleb/pytorch/torch/include/c10/core/Allocator.h:6,
                 from /home/gleb/pytorch/torch/include/ATen/ATen.h:7,
                 from /home/gleb/pytorch/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/gleb/pytorch/torch/include/torch/script.h:3,
                 from /home/gleb/GCNv2_SLAM/include/GCNextractor.h:24,
                 from /home/gleb/GCNv2_SLAM/include/Frame.h:32,
                 from /home/gleb/GCNv2_SLAM/include/MapPoint.h:25,
                 from /home/gleb/GCNv2_SLAM/include/KeyFrame.h:24,
                 from /home/gleb/GCNv2_SLAM/include/LoopClosing.h:24,
                 from /home/gleb/GCNv2_SLAM/src/LoopClosing.cc:21:

...

AndroidStudio2017, 2020-12-23:
This means you should use C++14 or later to compile the project.

Try to add commands below to your CMakeLists.txt
```
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
```

This should be work.

weihao-ysgs, 2021-05-01:
if you use QT,you can add some text in the .pro file as : CONFIG += c++14 console

HiIcy, 2022-10-26:
> This means you should use C++14 or later to compile the project.
> 
> Try to add commands below to your CMakeLists.txt
> 
> ```
> set(CMAKE_CXX_STANDARD 14)
> set(CMAKE_CXX_STANDARD_REQUIRED ON)
> ```
> 
> This should be work.

`set(CMAKE_CXX_STANDARD_REQUIRED ON)` this line help work!