jiexiong2016/GCNv2_SLAM #63: [libprotobuf FATAL /home/zhoulw/pytorch/third_party/protobuf/src/google/protobuf/generated_message_reflection.cc:2425] CHECK failed: file != nullptr: terminate called after throwing an instance of 'google::protobuf::FatalException

angiend, 2021-09-09:
os :ubunut18.04
python:2.7
gcc:7.5
protoc :3.0.0

and i have compiled the pytorch(libtorch) cpu version success,
and the GCNv2-slam also success,but run the "run.sh",an error happend,
what 's problem?

> zhoulw@ctqpc:~/slam/GCNv2_SLAM-master/GCN2$ ./run.sh 
> 
> ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: RGB-D
> 
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loaded!
> 
> 
> Camera Parameters: 
> - fx: 267.7
> - fy: 269.6
> - cx: 160.05
> - cy: 123.8
> - k1: 0
> - k2: 0
> - p1: 0
> - p2: 0
> - fps: 30
> - color order: RGB (ignored if grayscale)
> [libprotobuf FATAL /home/zhoulw/pytorch/third_party/protobuf/src/google/protobuf/generated_message_reflection.cc:2425] CHECK failed: file != nullptr: 
> terminate called after throwing an instance of 'google::protobuf::FatalException'
>   what():  CHECK failed: file != nullptr: 
> ./run.sh: line 10: 25719 Aborted                 (core dumped) GCN_PATH=/home/zhoulw/slam/GCNv2_SLAM-master/GCN2/gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml /home/zhoulw/slam/data/TUM/freiburg3/rgbd_dataset_freiburg3_long_office_household /home/zhoulw/slam/data/TUM//freiburg3/rgbd_dataset_freiburg3_long_office_household/associations.txt

angiend, 2021-09-09:
`#include <iostream>
#include <torch/script.h> // One-stop header.
#include <torch/torch.h>

using namespace std;
const char *net_fn = "/home/zhoulw/slam/GCNv2_SLAM-master/GCN2/gcn2_320x240.pt";
int main(){

    torch::Tensor tensor = torch::rand({2, 3});
    std::cout << tensor << std::endl;

    std::shared_ptr<torch::jit::script::Module> module;
    module = torch::jit::load(net_fn);
    if(module != nullptr)
	std::cout << "model load success" << std::endl;
}
`

i write a test code ,just load the pt file , also meet the error.
`zhoulw@ctqpc:~/zhoulw/test_libtorch_example/build$ cmake -DCMAKE_PREFIX_PATH=/home/zhoulw/pytorch/torch/lib/tmp_install ..
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found torch: /home/zhoulw/pytorch/torch/lib/tmp_install/lib/libtorch.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zhoulw/zhoulw/test_libtorch_example/build
zhoulw@ctqpc:~/zhoulw/test_libtorch_example/build$ cmake --build . --config Release
Scanning dependencies of target example-app
[ 50%] Building CXX object CMakeFiles/example-app.dir/example-app.cpp.o
[100%] Linking CXX executable example-app
[100%] Built target example-app
zhoulw@ctqpc:~/zhoulw/test_libtorch_example/build$ ls
CMakeCache.txt  CMakeFiles  cmake_install.cmake  example-app  Makefile
zhoulw@ctqpc:~/zhoulw/test_libtorch_example/build$ ./example-app 
 0.5977  0.0043  0.5706
 0.9369  0.7761  0.4529
[ Variable[CPUFloatType]{2,3} ]
[libprotobuf FATAL /home/zhoulw/pytorch/third_party/protobuf/src/google/protobuf/generated_message_reflection.cc:2425] CHECK failed: file != nullptr: 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: file != nullptr: 
Aborted (core dumped)
`

miRemid, 2022-10-26:
> I got some result: ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza. This program comes with ABSOLUTELY NO WARRANTY; This is free software, and you are welcome to redistribute it under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: RGB-D
> 
> Loading ORB Vocabulary. This could take a while... Vocabulary loaded!
> 
> Camera Parameters:
> 
> fx: 267.7 fy: 269.6 cx: 160.05 cy: 123.8 k1: 0 k2: 0 p1: 0 p2: 0 fps: 30 color order: RGB (ignored if grayscale)
> 
> But I got some errors: It say: what: Cannot initialize CUDA without ATen_cuda library. PyTorch splits its backend into two shared libraries: a CPU library and a CUDA library; this error has occurred because you are trying to use some CUDA functionality, but the CUDA library has not been loaded by the dynamic linker for some reason. The CUDA library MUST be loaded, EVEN IF you don't directly use any symbols from the CUDA library! One common culprit is a lack of -Wl,--no-as-needed in your link arguments; many dynamic linkers will delete dynamic library dependencies if you don't depend on any of their symbols. You can check if this has occurred by using ldd on your binary to see if there is a dependency on *_cuda.so library.
> 
> Can you help me?

i meet this problem too, did you solve this problem?

jf99, 2023-08-29:
I guess this happens if the employed pytorch version is incompatible with the installed cuda version. Both updating pytorch and downgrading cuda comes with other issues that need to be resolved. As a workaround, you can just use the CPU. Due to a poor design decision, the use of device "cuda:0" is hardcoded into the traced models (such as gcnv2_480x640.pt). Luckily, these pt files are really just zip archives. So you can unzip them, `sed -i 's/cuda:0/cpu/g'` the files in it and rezipt the whole thing. Worked for me, but now I still have other issues to solve.