jiexiong2016/GCNv2_SLAM #71: 训练代码会开源吗？

fox-007-source, 2022-09-13:


xiehousen, 2022-09-21:
How is your test result? My test results in ORB-SLAM3 are not as good as the original ORB.

007fox-bit, 2022-09-21:
Well，I haven't tested it due to some problem about installation. Can you reply your installation steps？

xiehousen, 2022-09-22:
ORB-SLAM? Or LibTroch？

007fox-bit, 2022-09-22:
> libtorch

007fox-bit, 2022-09-22:
> ORB-SLAM? Or LibTroch？

libtorch

xiehousen, 2022-09-22:
可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。

007fox-bit, 2022-09-22:
> 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
你用的是哪个版本，cuda的版本是哪个呢

xiehousen, 2022-10-09:
> > 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
> > 你用的是哪个版本，cuda的版本是哪个呢

CUDA 11.3, LibTorch就是和CUDA对应的一个版本

007fox-bit, 2022-10-09:
> > > 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
> > > 你用的是哪个版本，cuda的版本是哪个呢
> 
> CUDA 11.3, LibTorch就是和CUDA对应的一个版本。但是这个版本好像效果不好，你可以测一下

谢谢，这个问题前段时间已解决，目前一直在尝试复现训练代码，如果大佬有研究可以交流交流

xiehousen, 2022-10-10:
> > > > 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
> > > > 你用的是哪个版本，cuda的版本是哪个呢
> > 
> > 
> > CUDA 11.3, LibTorch就是和CUDA对应的一个版本。但是这个版本好像效果不好，你可以测一下
> 
> 谢谢，这个问题前段时间已解决，目前一直在尝试复现训练代码，如果大佬有研究可以交流交流

请问你是在ORB-SLAM3上复现的吗？效果如何呢？

007fox-bit, 2022-10-11:
> > > > > 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
> > > > > 你用的是哪个版本，cuda的版本是哪个呢
> > > 
> > > 
> > > CUDA 11.3, LibTorch就是和CUDA对应的一个版本。但是这个版本好像效果不好，你可以测一下
> > 
> > 
> > 谢谢，这个问题前段时间已解决，目前一直在尝试复现训练代码，如果大佬有研究可以交流交流
> 
> 请问你是在ORB-SLAM3上复现的吗？效果如何呢？

你好，我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果不好？

xiehousen, 2022-10-11:
> 我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果

是的，效果感觉还不如ORB特征点

007fox-bit, 2022-10-11:
> > 我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果
> 
> 是的，效果感觉还不如ORB特征点

我目前只是在orb-slam2上运行我感觉还行，orb-slam3和2在前端处理的策略上变化不大，作者在论文里给出的结果也不是最好的，所以应该很正常。您可以多试几组数据集看看。

miRemid, 2022-10-23:
> > > 可以在官网下载一个LibTorch的库，然后直接放到一个目录，编译的时候直接指定路径就可以了。
> > > 你用的是哪个版本，cuda的版本是哪个呢
> 
> CUDA 11.3, LibTorch就是和CUDA对应的一个版本。但是这个版本好像效果不好，你可以测一下

你好，我从官网下载了预编译的libpytroch并解压到/path/libtorch了，我在build.sh中将TORCH_PATH设置成了/path/libtorch但cmake时依然没有找到libpytorch，请问该如何设置？

```shell
TORCH_PATH set to: /path/libtorch
CMake Error at CMakeLists.txt:46 (find_package):
  By not providing "FindTorch.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Torch", but
  CMake did not find one.

  Could not find a package configuration file provided by "Torch" with any of
  the following names:

    TorchConfig.cmake
    torch-config.cmake

  Add the installation prefix of "Torch" to CMAKE_PREFIX_PATH or set
  "Torch_DIR" to a directory containing one of the above files.  If "Torch"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

miRemid, 2022-11-01:
> > > 我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果
> > 
> > 
> > 是的，效果感觉还不如ORB特征点
> 
> 我目前只是在orb-slam2上运行我感觉还行，orb-slam3和2在前端处理的策略上变化不大，作者在论文里给出的结果也不是最好的，所以应该很正常。您可以多试几组数据集看看。

你好再次打扰了，我现在用的libtorch11.6的版本，编译能够正常编译，但是无法运行，请问你遇到过这种情况吗？
```
❯ NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml /data/datasets/SLAM/TUM/rgbd_dataset_freiburg3_walking_halfsphere /data/datasets/SLAM/TUM/rgbd_dataset_freiburg3_walking_halfsphere/associations.txt

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters:
- fx: 267.7
- fy: 269.6
- cx: 160.05
- cy: 123.8
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)
terminate called after throwing an instance of 'c10::Error'
  what():  Legacy model format is not supported on mobile.
Exception raised from deserialize at ../torch/csrc/jit/serialization/import.cpp:273 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) + 0x6b (0x7f78c71682eb in /home/zengxy/Workspace/slam/libtorch/lib/libc10.so)
frame #1: c10::detail::torchCheckFail(char const*, char const*, unsigned int, char const*) + 0xd1 (0x7f78c7163e41 in /home/zengxy/Workspace/slam/libtorch/lib/libc10.so)
frame #2: <unknown function> + 0x3e0adb5 (0x7f78f61bedb5 in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
frame #3: torch::jit::load(std::shared_ptr<caffe2::serialize::ReadAdapterInterface>, c10::optional<c10::Device>, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) + 0x1cd (0x7f78f61c04dd in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
frame #4: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) + 0xe0 (0x7f78f61c30f0 in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
frame #5: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>) + 0x6f (0x7f78f61c329f in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
frame #6: ORB_SLAM2::GCNextractor::GCNextractor(int, float, int, int, int) + 0x653 (0x7f791e379403 in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
frame #7: ORB_SLAM2::Tracking::Tracking(ORB_SLAM2::System*, DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>*, ORB_SLAM2::FrameDrawer*, ORB_SLAM2::MapDrawer*, ORB_SLAM2::Map*, ORB_SLAM2::KeyFrameDatabase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) + 0x1f9f (0x7f791e3553ef in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
frame #8: ORB_SLAM2::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM2::System::eSensor, bool) + 0x5fd (0x7f791e33d45d in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
frame #9: main + 0x251 (0x55c9c9db4e21 in ./rgbd_gcn)
frame #10: __libc_start_main + 0xe7 (0x7f78c73d0c87 in /lib/x86_64-linux-gnu/libc.so.6)
frame #11: _start + 0x2a (0x55c9c9db660a in ./rgbd_gcn)

[1]    31731 abort (core dumped)  NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin
```

Baymax958, 2022-11-03:
> 

> 

你好，我也正在尝试复现训练代码，请问您有眉目了吗

007fox-bit, 2022-11-03:
> > 
> 
> > 
> 
> 你好，我也正在尝试复现训练代码，请问您有眉目了吗

你好，目前还没有，方便的话可以加个联系方式交流，qq是1726039332

Leaf-G, 2022-11-16:
> 
我也遇到了同样的情况 请问你解决了吗

Leaf-G, 2022-11-16:
> 

饭

Leaf-G, 2022-11-16:
> 你好再次打扰了，我现在用的libtorch11.6的版本，编译能够正常编译，但是无法运行，请问你遇到过这种情况吗？
>
> ```
> ❯ NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml [...]

我也遇到了同样的问题 请问你解决了吗

Crazylov3, 2023-02-02:
@007fox-bit Are you still working on this. I also try to re-train this model. However, I face with issue related to binary description training. Have you done yet?

Crazylov3, 2023-02-02:
> How is your test result? My test results in ORB-SLAM3 are not as good as the original ORB.

I believe you used 320x240 to run ORB-SLAM3 with GCNv2. If you do so, you should also use this resolution for original ORB for fairly comparation. In my results, GCNv2 still worker better than original ORB, at least in 320x240 version. Full resolution version (640x480) doesnt worker too well, I believe problem about the they didnt focus on training with high resolution image.

007fox-bit, 2023-02-22:
> @007fox-bit Are you still working on this. I also try to re-train this model. However, I face with issue related to binary description training. Have you done yet?

I also face the same problem.

Pepper-FlavoredChewingGum, 2023-04-01:
> 

> > 我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果
> 
> 是的，效果感觉还不如ORB特征点

请问您解决这个问题了吗？我也尝试在ORB-SLAM3上复现GCNv2，但是效果很差，甚至不如原版的GCN-SLAM。请问您对于这个问题有什么理解呢？

S1aoXuan, 2023-10-16:
> > > > > > 我是在尝试复现它的模型 训练代码，您目前是将模型作为orb-slam3的前端吗？在数据集上效果
> > > > > 
> > > > > 
> > > > > 是的，效果感觉还不如ORB特征点
> > > > 
> > > > 
> > > > 我目前只是在orb-slam2上运行我感觉还行，orb-slam3和2在前端处理的策略上变化不大，作者在论文里给出的结果也不是最好的，所以应该很正常。您可以多试几组数据集看看。
> > > 
> > > 
> > > 你好再次打扰了，我现在用的libtorch11.6的版本，编译能够正常编译，但是无法运行，请问你遇到过这种情况吗？
> > > ```
> > > ❯ NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml /data/datasets/SLAM/TUM/rgbd_dataset_freiburg3_walking_halfsphere /data/datasets/SLAM/TUM/rgbd_dataset_freiburg3_walking_halfsphere/associations.txt
> > > 
> > > ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
> > > This program comes with ABSOLUTELY NO WARRANTY;
> > > This is free software, and you are welcome to redistribute it
> > > under certain conditions. See LICENSE.txt.
> > > 
> > > Input sensor was set to: RGB-D
> > > 
> > > Loading ORB Vocabulary. This could take a while...
> > > Vocabulary loaded!
> > > 
> > > 
> > > Camera Parameters:
> > > - fx: 267.7
> > > - fy: 269.6
> > > - cx: 160.05
> > > - cy: 123.8
> > > - k1: 0
> > > - k2: 0
> > > - p1: 0
> > > - p2: 0
> > > - fps: 30
> > > - color order: RGB (ignored if grayscale)
> > > terminate called after throwing an instance of 'c10::Error'
> > >   what():  Legacy model format is not supported on mobile.
> > > Exception raised from deserialize at ../torch/csrc/jit/serialization/import.cpp:273 (most recent call first):
> > > frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) + 0x6b (0x7f78c71682eb in /home/zengxy/Workspace/slam/libtorch/lib/libc10.so)
> > > frame #1: c10::detail::torchCheckFail(char const*, char const*, unsigned int, char const*) + 0xd1 (0x7f78c7163e41 in /home/zengxy/Workspace/slam/libtorch/lib/libc10.so)
> > > frame #2: <unknown function> + 0x3e0adb5 (0x7f78f61bedb5 in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
> > > frame #3: torch::jit::load(std::shared_ptr<caffe2::serialize::ReadAdapterInterface>, c10::optional<c10::Device>, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) + 0x1cd (0x7f78f61c04dd in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
> > > frame #4: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) + 0xe0 (0x7f78f61c30f0 in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
> > > frame #5: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>) + 0x6f (0x7f78f61c329f in /home/zengxy/Workspace/slam/libtorch/lib/libtorch_cpu.so)
> > > frame #6: ORB_SLAM2::GCNextractor::GCNextractor(int, float, int, int, int) + 0x653 (0x7f791e379403 in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
> > > frame #7: ORB_SLAM2::Tracking::Tracking(ORB_SLAM2::System*, DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>*, ORB_SLAM2::FrameDrawer*, ORB_SLAM2::MapDrawer*, ORB_SLAM2::Map*, ORB_SLAM2::KeyFrameDatabase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) + 0x1f9f (0x7f791e3553ef in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
> > > frame #8: ORB_SLAM2::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM2::System::eSensor, bool) + 0x5fd (0x7f791e33d45d in /home/zengxy/Workspace/slam/GCNv2_SLAM/lib/libORB_SLAM2.so)
> > > frame #9: main + 0x251 (0x55c9c9db4e21 in ./rgbd_gcn)
> > > frame #10: __libc_start_main + 0xe7 (0x7f78c73d0c87 in /lib/x86_64-linux-gnu/libc.so.6)
> > > frame #11: _start + 0x2a (0x55c9c9db660a in ./rgbd_gcn)
> > > 
> > > [1]    31731 abort (core dumped)  NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin
> > > ```
> > 
> > 
> > 我也遇到了同样的问题 请问你解决了吗
> 
> 我也是

我也遇到了这个问题，毫无头绪，我用的torch版本式1.11.0 cuda版本是11.3

007fox-bit, 2024-03-19:
> > > 
> > 
> > 
> > > 
> > 
> > 
> > 你好，我也正在尝试复现训练代码，请问您有眉目了吗
> 
> 你好，目前还没有，方便的话可以加个联系方式交流，qq是1726039332

有很多小伙伴加我qq想交流gcnv2，但是我当时因为能力有限没能成功复现，因此很多问题难以回复。本着众人拾柴火焰高的思想，我建了一个qq群方便大家交流。群号是930373239

haithink, 2024-10-21:
> 你好再次打扰了，我现在用的libtorch11.6的版本，编译能够正常编译，但是无法运行，请问你遇到过这种情况吗？
>
> ```
> ❯ NN_ONLY=1 GCN_PATH=gcn2_320x240.pt ./rgbd_gcn ../Vocabulary/GCNvoc.bin TUM3_small.yaml [...]

使用libtorch-cxx11-abi-shared-with-deps-2.5.0+cu124也是同样的问题。