jiexiong2016/GCNv2_SLAM #72: Reproduce Training GCNv2

Crazylov3, 2023-02-02:
Do anyone try to re-train GCNv2? I try to train this model from scratch but I can reproduce the result. I have problem using straight-through estimator (which they mentioned in paper) to training with binary description, the loss does not converge Any suggests, pls

yhjsss, 2023-11-24:
Hello, I am learning relevant knowledge, but I encountered many problems when reproducing it. Can you share your training code? Thank you very much.

mono0826, 2024-04-02:
> Hello, I am learning relevant knowledge, but I encountered many problems when reproducing it. Can you share your training code? Thank you very much.

Hello,Did you reproduce the training code?